powsybl/powsybl-open-loadflow #31: Open Loadflow does not work with non impedant branches

murgeyseb, 2019-12-19:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
Currently, Open Loadflow does not support non impedant lines, transformers or dangling lines, that are accepted in IIDM.
It generates NaN values that are not monitored by the computation that returns a convergence flag.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
Import a network file.
Set the R and X value of any branch to 0.
Run the loadflow

All the result variables equals Double.NaN

* **What is the expected behavior?**
The results of the loadflow is OK.

* **What is the motivation / use case for changing the behavior?**
Handle correct computation on some input files

geofjamg, 2019-12-19:
#31 is a temporary workaround to avoid incorrectly supported data badly propagate to computation core.
So now we need to decide how to fix this. I can see 2 solutions:
 - Like hades2, cut admittance to a minimal value. It is easy to develop but it might change a little bit the final computed state.
 - Handle non impedant branches in the equation system, with something like v = v and theta = theta between the 2 buses of the  non impedant branch.

annetill, 2019-12-19:
Buses of a non impedant branch have to be aggregated in the mathematical resolution I think. But the idea of cutting non impedant branch to low impedant branch could be a fast but temporary solution. 

But it is possible that we also need to have a default reactance value to deal with too low impedant lines. Then, we have to choose between 2 strategies:
- First, we increase low reactance to the default value and we compute low impedance lines as lines.
- Second, more complex, we decide to remove low impedance lines and we agregate the buses for the resolution. But we need a graph post-processing to compute flows on these low impedance lines. 

Just for information: by default HELM-Flow considers that a branch has "low impedance" if its reactance is below 0.00029 in per-unit. This is also the default value used in GE PSLF (a power flow tool used by many companies in the US) and established by the WECC (Western Electricity Coordinating Council) in its data preparation manual.

annetill, 2019-12-19:
Just a question: for transformer with null or low impedance, we have to increase the reactance to the default value... no ?

closed 2020-04-08 by geofjamg