powsybl/powsybl-open-loadflow #45: Shared local control for generators not working

annetill, 2020-02-06:
* **Do you want to request a *feature* or report a *bug*?**

It could be considered as a bug.

* **What is the current behavior?**

It seems that a shared control at local common local bus for generators is not working when the parameter "GeneratorVoltageRemoteControl" is active. 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

The use case is the one used in tutorial "cgmes". The open load flow parameters should be:
OpenLoadFlowParameters openLoadFlowParameters = new OpenLoadFlowParameters().setGeneratorVoltageRemoteControl(true);

* **What is the expected behavior?**

If we set:
````
networkBe.getGenerator("_3a3b27be-b18b-4385-b557-6735d733baf0").setVoltageRegulatorOn(false);
```` 
we should obtain the same results as if 
````
OpenLoadFlowParameters openLoadFlowParameters = new OpenLoadFlowParameters().setGeneratorVoltageRemoteControl(false);
````
but it is not the case.

* **What is the motivation / use case for changing the behavior?**

Huge differences in that case.

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2020-02-06:
setGeneratorVoltageRemoteControl on open load flow parameters only configure the way we manage remote voltage, either with a scaling or with additional equations to the equation system.
setVoltageRegulatorOn on a generator is a different thing, it only configure if one particular generator control the voltage or not (PV/PQ bus).

annetill, 2020-02-06:
Yes in fact to be more precise : the generator I propose to put out of regulation is the only one in the study network that could controls a remote point. It is a way to say that if we don't have this generator in the newtork, the open load flow option "setGeneratorVoltageRemoteControl" set to true or false should provide the same results.

geofjamg, 2020-02-06:
You can't have the same result because you won't have the same number of PV bus (and it is not linked to the way we handle remote voltage, with scaling or not).

closed 2020-02-10 by geofjamg