powsybl/powsybl-open-loadflow #84: Phase control could be improved if needed

annetill, 2020-05-12:
* **Do you want to request a *feature* or report a *bug*?**

It is more a feature.

* **What is the current behavior?**

We have recently merged the control of active power by phase tap changer. This control can be located at side 1 or at side 2 of the branch, where the tap changer is. As a beta version, we only support two windings transformers. Furthermore, the active power target is often given with a deadband. The open loadflow logs if the active power at regulation terminal is in or out the deadband.  

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**

There is no expected behavior. Phase control in this open loadflow is a minimal viable feature and could be improved. I am writing this issue to let some notes of what could be missing.

* **What is the motivation / use case for changing the behavior?**

- The deadband could be taken into account to select the rounded step after the continuious resolution. In fact, we have to choose between 2 steps. In the current implementation, we choose the step which alpha is the closest of the continuious value. 
- The regulating terminal could be remote ;
- We don't support phase control on phase tap changer belonging to three windings transformers ;
- Maybe a connexity computation to evaluate the efficiency of the phase control ;
- Phase control for active power flow (DC mode) ;
- Phase control for phase tap changer which control mode is CURRENT_LIMITER (https://www.powsybl.org/docs/iidm/model/phaseTapChanger.html). The continuous problem is solved only for tap changers that are out of its regulation value.

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

If other features are missing, please leave a message in this discussion.

annetill, 2020-11-20:
We have started to work on current limiter mode in a branch: https://github.com/powsybl/powsybl-open-loadflow/tree/current-limiter-mode
To be finished.

geofjamg, 2022-03-13:
Solved by #340

closed 2022-03-13 by geofjamg