powsybl/powsybl-open-loadflow #97: Static var compensator voltage control increasing

annetill, 2020-05-20:
* **Do you want to request a *feature* or report a *bug*?**

A feature. 

* **What is the current behavior?**

We model the voltage remote control of static var compensator as we do with generators with PQV nodes and equations that link the reactive power of controllers (2 by 2). 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

It is not a bug but sometimes we prefer to have an simplified approach. 

* **What is the expected behavior?**

````When the cim:StaticVarCompensator.slope is positive (and present in an IGM CIM-CGMES file), the reactive output of the SVC is defined by:
Q=(V_bus-V_ref)/slope   
Where:
	cim:StaticVarCompensator.slope is positive based on the load sign convention.
	Vbus is the voltage at the controlled point determined by cim:RegulatingControl.Terminal which may be local or remote.
	Vref is the target value given by cim:RegulatingControl.targetValue.````

Extract from the ENTSO-E implementation guide for CGM data modeling 1.0

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2020-05-21:
Such equipment may also have reactive power limits?

annetill, 2020-11-04:
[image]

This model could be useful. 

The generator model represents the slope by connecting ths SVC to an auxiliary bus separated from the high voltage bus by a reactance equal to the per unit slope. When the SVC is operating the limits, its behaves like a fixed susceptance and its injected reactive power is proportional to the square of the regulated bus voltage.

geofjamg, 2022-03-13:
Done in #391

closed 2022-03-13 by geofjamg