powsybl/powsybl-open-loadflow #110: Increase the slack distribution feature

annetill, 2020-06-05:
* **Do you want to request a *feature* or report a *bug*?**

A feature.

* **What is the current behavior?**

Distributed slack is supported in the open-loadflow considering iidm:Generator.MaxP and iidm:Load.TargetP.  

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**

We can imagine that the slack could be distributed in separating loads that follow a daily load pattern from the others. The first kind of loads are cim:ConformLoad and the pattern can be use to scale the load. 

When scaling these load, the power factor (=cos(arctan(Q/P)) has to remain a constant value. 

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

annetill, 2020-06-26:
This feature needs an IIDM evolution https://github.com/powsybl/powsybl-core/issues/1369

annetill, 2020-07-09:
You have to use the ````LoadDetail```` extension:
```java
LoadDetail loadDetail = load.getExtension(LoadDetail.class)
if (loadDetail != nnull) {
            peFixe = checkValue(detail.getFixedActivePower());
            peAff = checkValue(detail.getVariableActivePower());
            qeFixe = checkValue(detail.getFixedReactivePower());
            qeAff = checkValue(detail.getVariableReactivePower());
}
```

annetill, 2020-10-13:
This issue is partially solved by adding BalanceType.PROPORTIONAL_TO_CONFORM_LOAD support. It remains the power factor to be kept constant.

frigaux, 2020-11-13:
Report about phone call with Anne on 11/13/2020

1. add a new boolean "isPowerFactorConstant" in OpenLoadFlowParameters
2. implements constant power factor in "DistributedSlackOnLoadOuterLoop:run". Power factor equation is given in [LCC Converter Station](https://www.powsybl.org/pages/documentation/grid/model/) and must be constant if "isPowerFactorConstant" is true
3. check metrics in DistributedSlackOnLoadTest : iteration count must not increase with a new feature, how to check runtime not increased too ?

frigaux, 2020-11-24:
Report : teams call with Anne on 11/20/2020 about missing feature :

We have to handle two case : `balanceType` = `PROPORTIONAL_TO_LOAD` and `balanceType` = `PROPORTIONAL_TO_CONFORM_LOAD`

- `balanceType` = `PROPORTIONAL_TO_LOAD`, case described in comment dated 11/20/2020

- `balanceType` = `PROPORTIONAL_TO_CONFORM_LOAD`, in this case, we have to keep powerFactor constant on **variable power only**, it means we have to isolate Q_2 in this equation :

>$$
QV_2={\frac {{QV_1}{PV_2}} {PV_1}}
$$

>$$
Q_2-QF={\frac {{(Q_1-QF)}{(P_2-PF)}} {P_1-PF}}
$$

>$$
Q_2=QF+{\frac {{(Q_1-QF)}{(P_2-PF)}} {P_1-PF}}
$$

AnkurArohi, 2022-02-23:
I have some questions regarding the approach, I find it interesting, the first question is regarding the basics of power flow , 

if we have a distributed slack feature, does it mean that all the buses which have to be considered as slack bus will have a delta(voltage angle) zero as is normally with a Slack bus?

Secondly, if I have understood it correctly this would just distribute the load Power requirement amongst the slacks,like a redistribution of Active power requirement keeping the load power factor same, this seems to be more of a distributed generation case (Redispatching P) that a distributed slack, which would further mean that the modelled buses are of PU type and not UDEL type, s this correct?

Thirdly, the redistribution of Q from the equations above respecting the constraint of constant power factor is actually a power flow problem, as it depends also on the voltage profiles, I think the equational description as above is simplifying it a lot, I would actually let the power flow determine the Q values after the NR convergence iteration keeping of course the constraint of constant power factor at the conform loads. So basically the modelling of the distributed generation will be a PV bus where in the delta and Q values will be calculated from NR but the P value will be provided from the redistribution of the load power requirement (Redispatching P), Is that not so?

I would appreciate your expert advice and comments

annetill, 2022-02-23:
> I have some questions regarding the approach, I find it interesting, the first question is regarding the basics of power flow ,
>
> if we have a distributed slack feature, does it mean that all the [...]

It is not our approach here: we need of course one slack bus per synchronous component, in order to put the mismatch after each NR. But we have an outer loop to distribute the global mismatch on equipments. The most common way is to increase or decrease generators production. But in some cases, we distribute the global slack on loads (we increase or decrease the consumption).

> Secondly, if I have understood it correctly this would just distribute the load Power requirement amongst the slacks,like a redistribution of Active power requirement keeping the load power factor [...]

With OLF, the power factor could remains constant only on loads. On generator, this feature has no sense. Indeed this issue is linked to a requirement of the european merging function (https://eepublicdownloads.entsoe.eu/clean-documents/Network%20codes%20documents/Implementation/cacm/cgmm/European_Merging_Function_Requirements_Specification.pdf).

> Thirdly, the redistribution of Q from the equations above respecting the constraint of constant power factor is actually a power flow problem, as it depends also on the voltage profiles, I think the [...]

I think it is what we are doing, my issue was to simple. 

> I would appreciate your expert advice and comments

AnkurArohi, 2022-02-23:
> It is not our approach here: we need of course one slack bus per synchronous component, in order to put the mismatch after each NR. But we have an outer loop to distribute the global mismatch on [...]
Thankyou for the reply, so are you distributing it on loads/increasing-decreasing the load ? I mean I fail to understand the relationship between the Distributed Slack feature and Loads...

> With OLF, the power factor could remains constant only on loads. On generator, this feature has no sense. Indeed this issue is linked to a requirement of the european merging function [...]

AnkurArohi, 2022-02-23:
If the feature is called distributed slack then It should have nothing to do with manipulating the load profiles, rather only with slack, isn't it? plus can we consider multiple slacks in a classical NR problem?