powsybl/powsybl-open-loadflow #129: Fast security analysis

geofjamg, 2020-07-16:
We need to implement a security analysis much faster than default one provided by `com.powsybl.security.SecurityAnalysisImpl` which is an external loop over `com.powsybl.loadflow.LoadFlow` run.

annetill, 2020-10-13:
This issue is partially solved by this https://github.com/powsybl/powsybl-open-loadflow/pull/135

annetill, 2020-12-21:
The DC mode of the fast security analysis would be more efficiently done through our new `OpenSensitivityAnalysisProvider`. This mode could support actions too.

annetill, 2021-03-24:
We still have a beta version of the security analysis. What we still need to support is:

- Equipements as contingency that are not branches: loads, generators, static var compensators, VSC, shunt compensators, bus bar sections, HVDC lines. The HVDC contingency is not done correctly as we don't model the physical link between the two converter stations. I have started a small prototype for that based on variable equation terms.
- Apparent power limits: permanent and temporary ;
- To remove from slack distribution the loads and the generators that are disconnected through the contingency ;
- Contingency that separates controllers buses/branches and controlled bus/branch ;
- Contingency that separates shared controllers (see CGMES conformity test cases) ;

annetill, 2021-05-21:
@obrix and @Hadrien-Godard can you handle the two first bullets points?
- For all `ContingencyElementType` support, I think that you can start with `GENERATOR` and `SVC`.
- For all kind of limits: the apparent power limits and active power limits. You can start with `ActivePowerLimits` (in MW) and continue with `ApparentPowerLimits` (in MVA).

geofjamg, 2022-03-13:
I close this issue because most of it has been done and because it is too generic. Additional issues will be created to improve security analysis.

closed 2022-03-13 by geofjamg