powsybl/powsybl-open-loadflow #160: Improve performances of security analysis

annetill, 2020-10-13:
* **Do you want to request a *feature* or report a *bug*?**

A feature.

* **What is the current behavior?**

In the beta version of the security analysis https://github.com/powsybl/powsybl-open-loadflow/pull/135, the jacobian matrix is rebuild before each post contingency simulation. But, between 2 contingencies, most of the time, no equation have been descativated and the rebuild is not necessary. 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**

I think that as a first approach, the jacobian matrix has to be rebuild only if an equation is desactivated. It is not the best way to handle performance issue, but maybe the easiest one. It could be interesting to sort the contingency list by activePowerLoss or to find a criteria that minimizes the number of desactivated equations after a post contingency simulation. 

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

floriand-e2r, 2020-10-13:
Besides the jacobian matrix, some pretty minor improvements:
- the state vector can be kept instead of being rebuild each time: in the Newton-Raphson the voltage initializer only needs to be used at iteration=0
- the equation system does not need to be updated with the state vector within the outerloops
- the equation vector does not need to be reset to 0 as all its values are updated

geofjamg, 2022-03-13:
This is already how it works

closed 2022-03-13 by geofjamg