powsybl/powsybl-open-loadflow #186: Manage Contingencies in Sensitivity Analysis

Djazouli, 2020-12-11:
* **Do you want to request a *feature* or report a *bug*?**
A feature

* **What is the current behavior?**
Currently, if a ContingenciesProvider is supplied to the OpenSensitivityAnalysisProvider.run method, an UnsupportedOperationException is raised.

* **What is the expected behavior?**
We should be able to supply sensitivity values for each contingency.

annetill, 2020-12-11:
**Use case**
	- Sensitivity of +1 MW on a bus n on the flow of the line (i,j) with the contingency of the line (m,k)
	- In a first approach, we will not support phase shifter at line (m, k)
	
**Right hand side B**
	- First vector with +1 MW at bus n
	- Second vector with +1 MW at bus m and -1 MW at bus k
	         - If bus m is the slack bus, the second vector is only +1 MW at bus k
	         - If bus k is the slack bus, the second vector is only -1 MW at bus m
	
We solve the system JT = B from the LU factorisation to obtain T, the state vector of angles. Then:
	
1) Computation of sensitivity value of bus n on line (i,j) (already exists):
	Snij = (Ti1 - Tj1)/Xij
	
2) Computation of sensitivity value N-1 of bus n one line (i,j) with a line contingency:
	Snijmk = Snij + (Tm1-Tk1)/(Xmk - (Tm2 - Tk2)) * (Ti2-Tj2)/Xij

Djazouli, 2020-12-22:
The 2. mentioned above comes from the inversion of a linear system. This method will always work and give the right result, except if the system cannot be inverted. This can only happen if adding +1-1 on the buses of the line (m,k) produce a flow of 1 on this line. This can only happen if there no other path between m and k than this branch. So, the line (m,k) is a "bridge" between two parts of the network, let's call them A and B. The slack bus is in A with m, and k is in B.

In this case, adding a +1-1 on m and k cannot change the flow on the branch (i,j) that we are watching.
So, our N-1 sensitivity can only be equal to the N sensitivity, if the N-1 sensitivity makes sense.

Let's call b the bus where we are making an injection. 

If b is in A, then the aforementioned system becomes 0=0, so, even if it is not invertible, we can solve it, and the N-1 sensitivity makes sense.

If b is in , the system becomes 0=1, and we cannot solve it. We then have two cases:

- If (i,j) was in A, the sensitivity is obviously 0.
- If (i,j) was in B, the sensitivity has no sense because there is no slack in B. But if we consider that the slack of B is k, then the sensitivity N-1 will be equal to the sensitivity N, because k represents the entrance to the "bridge", the only path available to the old slack.

closed 2021-01-26 by Djazouli