powsybl/powsybl-open-loadflow #188: Voltage magnitude is undefined

mathbagu, 2020-12-16:
* **Do you want to request a *feature* or report a *bug*?**
Bug?

* **What is the current behavior?**
On several of my old real cases, I have an exception "Voltage magnitude is undefined for bus XXX".

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
The bug can be reproduce with open-loadflow v0.8.0 on these cases:
- 20191204_1030_2D3_D20.zip (CGMES)
- 20190821T0830Z_1D_NG__004.zip (CGMES)
- 20151011_1030_FO7_FR0.uct (UCTE)

* **What is the expected behavior?**
I used to run load-flow on these files, so I suppose there could be a regression somewhere

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: 3.8.0
  - OS Version: 0.8.0

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2020-12-16:
I guess you have voltage init mode in your parameters to previous value?
If you don t have state var in your case you have to use initial value mode.

mathbagu, 2020-12-16:
My bad, this parameter is override by my configuration file. I wonder if we should be robust and automatically switch to initial value or implement a test with a more functional message (Impossible to run load-flow in "previous_value" mode because some voltage magnitudes are missing" or something like that

What do you think? If it's not relevant, feel free to  close this issue

geofjamg, 2020-12-16:
> My bad, this parameter is override by my configuration file. I wonder if we should be robust and automatically switch to initial value or implement a test with a more functional message (Impossible [...]

Indeed it would make sense to just warn if at least one voltage is missing and fallback to flat start.

geofjamg, 2022-03-13:
I close this issue because there is another one similar to think about this topic #429

closed 2022-03-13 by geofjamg