powsybl/powsybl-open-loadflow #201: Apparent power limits support

annetill, 2021-01-20:
* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

annetill, 2021-06-17:
Solved partially with #313, missing for DC sensitivity analysis using `cos phi`.

geofjamg, 2022-03-13:
> Solved partially with #313, missing for DC sensitivity analysis using `cos phi`.

What is the link with DC sensitivity analysis?

annetill, 2022-03-15:
> > Solved partially with #313, missing for DC sensitivity analysis using `cos phi`.
> 
> What is the link with DC sensitivity analysis?

Because if the network doesn't have any active power limits, but current limits only, it does work. So we can imagine to convert active power flows in currents to increase DC security analysis (and not DC sensitivity analysis sorry).

geofjamg, 2022-03-15:
> Because if the network doesn't have any active power limits, but current limits only, it does work. So we can imagine to convert active power flows in currents to increase DC security analysis (and [...]

Yes, right, we should create a more specific issue for this.

geofjamg, 2022-08-23:
So thanks to #549, this issue seems to be fully fixed.

closed 2022-08-23 by geofjamg