powsybl/powsybl-open-loadflow #218: AbstractLfBus.dispatchQ : sum of generators calculatedQ = qToDispatch ?

frigaux, 2021-02-08:
* **Do you want to request a *feature* or report a *bug*?**
*bug* ?

* **What is the current behavior?**
dispatch Q between StaticVarCompensator with broken limits seems to be wrong ?

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
Given Network :
  - Bus vl1_0 from VoltageLevel vl1 with NominalVoltage = 400.0 
    Generator bus1gen with : TargetP = 101.36639999999998 ; MinP = 0.0 ; MaxP = 150.0 ; VoltageRegulatorOn = true ; TargetQ = NaN
    Line bus1bus2line with : bus1 = vl1_0, bus2 = vl2_0, R = 1.0, X = 3.0, G1 = 0.0, G2 = 0.0, B1 = 0.0, B2 = 0.0
  - Bus vl2_0 from VoltageLevel vl2 with NominalVoltage = 400.0 
    Static Var Compensator bus2svc2 with : Bmin = -0.006 ; Bmax = 0.006 ; VoltageRegulatorOn = true ; VoltageSetpoint = 385.0 : Slope = 0.01
    Static Var Compensator bus2svc3 with : Bmin = -0.001 ; Bmax = 0.001 ; VoltageRegulatorOn = true ; VoltageSetpoint = 385.0 : Slope = 0.015
    Static Var Compensator bus2svc4 with : Bmin = -7.5E-4 ; Bmax = 7.5E-4 ; VoltageRegulatorOn = true ; VoltageSetpoint = 385.0 : Slope = 0.02
    Line bus1bus2line with : bus1 = vl1_0, bus2 = vl2_0, R = 1.0, X = 3.0, G1 = 0.0, G2 = 0.0, B1 = 0.0, B2 = 0.0

AbstractLfBus.dispatchQ is called with with qToDispatch = *-6.412103131789854*
at the end, we get generators with these calculatedQ :
  - bus2svc2 : calculatedQ = -4.8868494870881636
  - bus2svc3 : calculatedQ = -1.48225
  - bus2svc4 : calculatedQ = -1.1116874999999999
Sum of generators calculatedQ = *-7.48* <> qToDispatch ?

* **What is the expected behavior?**
Sum of generators calculatedQ = qToDispatch ?

* **What is the motivation / use case for changing the behavior?**
working on new bus PVLQ with many StaticVarCompensators and limits broken

* **Please tell us about your environment:**
  - powsybl-open-loadflow: 0.9.0-SNAPSHOT
  - OS Version: windows 10 pro + jdk1.8.0_271

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

annetill, 2021-02-12:
Can you provide a test case that shows the bug please?

frigaux, 2021-02-12:
Just push a junit test that fails with network given in this issue description.
You can find it in branch "issue_218_dispatchQ" :
https://github.com/powsybl/powsybl-open-loadflow/tree/issue_218_dispatchQ

Current build of this branch ends with this failure :
*Error:  Failures: 
Error:    LfBusImplTest.updateGeneratorsStateTest:93 sum of generators calculatedQ should be equals to qToDispatch ==> expected: <-6.412103131789854> but was: <-7.4807869870881625>*

closed 2021-03-12 by annetill