powsybl/powsybl-open-loadflow #258: Criterium to prevent cases where two PV buses have a different target and are linked with low impedent lines

Hadrien-Godard, 2021-03-24:
* **Do you want to request a *feature* or report a *bug*?**
A feature.

* **What is the current behavior?**
KLU issues may occur when an AC loadflow is performed on a network where a pair of PV buses linked by low impedent lines have a different voltage target.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
It is not a bug.

* **What is the expected behavior?**
The expected behavior is to not launch the loadflow before having remove a voltage controler from the regulation (beginning with transformers and ending with generators).
To detect this case, the following algorithm is proposed:
1/ Detect the pairs of PV buses which are at the same voltage level and have different voltage targets,
2/ Build the shortest path in term of line impedance between both PV buses.
3/ If the length of the path exceeds V1*(V1-V2)/Qmax, there is no risk to perform the loadflow with both buses as PV-buses.
4/ If the length of the path is under the criterium (V1*(V1-V2)/Qmax), one of the PV bus must be switched to a PQ-bus (transformers before genrators).
V1 and V2 are the two voltage targets, V1 is the larger target. Qmax depends on the voltage level:
100 kV: 400 MVar
220 KV: 880 Mvar
400 kV: 1600 MVar
750 kV : 3000 MVar

* **What is the motivation / use case for changing the behavior?**
Avoiding crashes of loadflow runs.

annetill, 2021-04-27:
I have tried to use this new feature on real test cases and it seems that we still have KLU issue. Furthermore, it does not solve the controlled buses that have a shared mixed control (generators + ratio tap changers), but for sure it was not the purpose of this PR but it makes me thinking about a new (old) approach of resolving generation and tap changers regulations in two step:
- Simulating first the generation controls ;
- For the generator regulating terminals close or common to tap changer regulating terminals, we switch off the voltage regulation and we model the voltage regulation of tap changers. For that, we have to increase the outer loops by an `init()` method that will change the network after loading (in deactivating the tap changer controls for example) and change the tap changer control outer loop with a small graphe analysis.

AnkurArohi, 2022-05-18:
@Hadrien-Godard 
I understand 👍 
But this is not the case for low voltage grid abd people would like to use powsybl also in there so to accept a global solution I would still suggest adding a resistance of negligible value or a virtual switch

vidaldid-rte, 2025-03-12:
There are two developments related this issue
  - now that #1135 is merged, the problem of KLU is largely reduced.
There are still cases where inconsistent voltage can cause convergence issues. We have a #1115 currently under evaluation.

If you have feedback on the remaining frequency of KLU or the interest in #1115 you can submit it here.