powsybl/powsybl-open-loadflow #289: Contingency propagation in sensitivity analysis

sylvlecl, 2021-04-26:
* **Do you want to request a *feature* or report a *bug*?**

Bug.

* **What is the current behavior?**

Sensitivity analysis is performed without keeping switches which are necessary for proper propagation of some contingencies.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

TODO

* **What is the expected behavior?**

Contingencies should correctly propagate in sensitivity analysis, opening all relevant switches. 
An issue seems to be that fast DC sensitivity analysis cannot handle, for now, that kind of contingencies.
Therefore, for DC analysis, we could either:
* throw if such a contingency is defined in inputs
* not propagate that kind of contingencies, but we should have a warning / doc about this

* **Please tell us about your environment:**
  - PowSyBl Version: powsybl-open-loadflow 0.8.0-SNAPSHOT

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

In version 0.7.0, propagated contingencies ended with an exception:
```
The contingency on the branch <SWITCH ID> not found in the network
```
Since commit https://github.com/powsybl/powsybl-open-loadflow/commit/32b3b5fc15589af485d9b970e95dda5b1d2b8a82, there is not more exception, but the behaviour is not right: 
switches are just skipped, they stay closed whereas they should open.

It would make sense, at least for AC sensitivity analysis, to keep breakers in order to be able to properly propagate contingencies, like in AC security analysis.

annetill, 2021-04-27:
Thanks a lot @sylvlecl for this well described issue. The code to solve this issue could be quite consequent as all the sensitivity analysis are performed on the bus view. We made this choice for performance issue as a first approach. But, you are right this kind of propagation is not well handled. I propose to solve this issue in two steps:
- A warning for these kind of contingency, that will not be propagated at all and a documentation ;
- Try to stay in bus view in our sensitivity analysis: it means that we have to improve our pre processing on the bus breaker view to detect all the equipments that will be lost: lines, buses. For buses that will be separated in several buses after a contingency, we have to list what we lost: loads, generators, dangling lines, shunts, injections in general and change the RHS in consequence. But it is really ambitious. If this approach is really costly (because for the moment, we only support branch contingency in the simulator), we have to think about plugging the AC sensitivity analysis on the bus breaker view. For the DC sensitivity analysis, as it is really simple by definition, it is really a need? @Hadrien-Godard what is your opinion about that?

Hadrien-Godard, 2021-04-28:
I agree with @annetill on how to handle this issue. This seems to me to be the right approach.

annetill, 2021-04-30:
https://github.com/powsybl/powsybl-open-loadflow/commit/cf20b9d5a1b6a6a7d82db697a343c8a5e2880d9d