powsybl/powsybl-open-loadflow #290: Support of DanglingLine contingency

annetill, 2021-04-27:
* **Do you want to request a *feature* or report a *bug*?**

Maybe a bug.

* **What is the current behavior?**

We don't support DanglingLine contingency in our sensitivity analysis and in our security analysis. I don't know exactly the be code behavior but it would be easy to make small unit tests in each module to see.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

Probably.

* **What is the expected behavior?**

It is easy to support it.

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

annetill, 2021-06-17:
Solved for DC sensitivity only with #300

geofjamg, 2022-03-13:
Also supported in AC security analysis and AC sensitivity analysis

closed 2022-03-13 by geofjamg