powsybl/powsybl-open-loadflow #296: [Sensitivity] Performance improvement of input reading for voltage sensitivity analysis

sylvlecl, 2021-04-29:
* **Do you want to request a *feature* or report a *bug*?**

Perf improvement

* **What is the current behavior?**

Buses for which we want to compute the voltage are retrieved by their ID, which is very inefficient in IIDM implementation for node breaker networks (since no index of calculated buses is maintained).
For a 200 buses x 3000 generators computation, most of the time seems to be spent in that part, not in the actual computation:
```
11:07:24.434 [System-8] INFO  c.p.openloadflow.network.LfNetwork - Network 0 balance: active generation=125576.9315011931 MW, active load=123155.10040852637 MW, reactive generation=-5120.065235338028 MVar, reactive load=25424.730541984783 MVar
11:07:40.429 [System-8] INFO  c.p.o.s.AbstractSensitivityAnalysis - Running AC sensitivity analysis with 645600 factors and 0 contingencies
```

* **What is the expected behavior?**

A negligible part of the computation should be spent in building those inputs.

* **What is the motivation / use case for changing the behavior?**

Performance for large networks.

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

Possible solutions:
* maintain in OLF a mapping bus ID --> Bus which can be reused anywhere
* evolution in IIDM implementation in powsybl-core to maintain this kind of mapping in the network
* change the way inputs are defined for voltage sensitivity (not a raw bus ID)

sylvlecl, 2021-04-29:
A easy and quick win is implemented in #298 : buses are cached so that they are not searched 3000 times if there are 3000 generators. It allows to reduce computation time from ~25s to ~8s on the use case.

However even for one call, the search remains inefficient : still to be improved.

geofjamg, 2022-03-13:
@annetill @sylvlecl I would close this issue, because mostly solved with bus cache and more a IIDM issue. Agreed?

sylvlecl, 2022-03-14:
Indeed!

closed 2022-03-14 by sylvlecl