powsybl/powsybl-open-loadflow #417: Repository reorganization

geofjamg, 2022-01-09:
* **Do you want to request a *feature* or report a *bug*?**
feature

* **What is the current behavior?**

In order to maybe add additional simulators in the repository, we could reorganize the repo.

* **What is the expected behavior?**

powsybl-openloadflow repository would be renamed powsybl-open-simulator
It would be a multi-module Maven project with the following sub modules:
powsybl-open-simulator
  - powsybl-open-simulator-matrix (transfer of com.powsybl.math.matrix from powsybl core)
  - powsybl-open-simulator-network (would contain com.powsybl.openloadflow.network, LfNetwork renamed OsNetwork)
  - powsybl-open-simulator-core (would contain com.powsybl.openloadflow.equation)
  - powsybl-open-loadflow (remaining code from actual repo: lf + as + sensi, DC + AC)

* **What is the motivation / use case for changing the behavior?**
Extend OLF repo scope and having a more modular and extendable code.

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

annetill, 2022-01-09:
Yes I trust you on that, I have no skill to resolve this issue :-) Why `OsNetwork` ? For the transfer of the matrix management, I can find someone from my team.

Maybe, before doing that, it would be great to merge all the PRs open that are close to be finish. What do you think ?

geofjamg, 2022-01-10:
> Why `OsNetwork` ?

Like Open Simulator Network

geofjamg, 2022-01-10:
> Maybe, before doing that, it would be great to merge all the PRs open that are close to be finish. What do you think ?

Sure