powsybl/powsybl-open-loadflow #429: Behaviour of initialization on previous values

sylvlecl, 2022-01-18:
* **Do you want to request a *feature* or report a *bug*?**

Ambiguous.

* **What is the current behavior?**

When using `PREVIOUS_VALUES` as initialization option, we may get an exception with following message:
```
Voltage magnitude is undefined for bus '****'
```

This can make that option hard to use, because it can happen even after a very minor topology change. 

* **What is the expected behavior?**

To be discussed: we might want to fill the holes when they occur ?
Or maybe we could fall back on a DC values initialization ?
Alternatively, maybe this issue must be left for the user to handle, through a try/catch for example, or ensuring that he uses that option only when he is sure to have results on all buses.

geofjamg, 2022-03-13:
Similar issue than one described in #188

annetill, 2022-10-18:
In case of any operator strategy in the security analysis, we have added this fall back (https://github.com/powsybl/powsybl-open-loadflow/blob/main/src/main/java/com/powsybl/openloadflow/network/util/PreviousValueVoltageInitializer.java). We can discuss now added this for the contingency state computation. @sylvlecl what do you think?

vidaldid-rte, 2025-03-12:
Somme feedback:

With the view that PreviousValueVoltageInitializer should allow to reproduce a Security Analysis interactively (run network, disconnect something, run again from previous equilibrium, make a corrective action, run again from previous equilibrium) then it makes sense to expose the fallback parameter.
I is also good to keep the strict mode (error) for example to detect cases where the previous run was not run all newtork components.

benoitflo, 2025-03-18:
Another feedback :
I am using pypowsybl with many iterative loadflow. I will appreciate to use init voltage mode with  previous values in a permissive way. One idea could be to replace missing value by default values (1pu, 0rad for instance). I generally have one or to bus without values, in case a check over the number of default value should be monitored.
Thanks