powsybl/powsybl-open-loadflow #456: OpenLoadflow ignores all AC Line segments not defined in BD File(DanglingLines) while importing Microgrid CGMES Files for LoadFlow

AnkurArohi, 2022-02-23:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
While preparing the Network object for LoadFlow the AC Linesegments are interpreted as Dangline Lines and for this purpose the DanglingLine_Bus objects are created.
So for example if you have a network with AC Line segments, the AC Line segments which are not Dangling Lines as per the BD File, would not be forming the required bus and hence would be out of the Network Object leading to a Nan value calculation for current even if the Power Flow converges

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
Microgrid CGMES File Import in Debug mode before the Network is ready for power Flow calculation

* **What is the expected behavior?**

All AC Line Segments to be part of Network ready to be used for Power Flow calculation, even if they are not dangling lines

* **What is the motivation / use case for changing the behavior?**
Al AC Line Segments to be imported and mapped in the Network object for the Power Flow Calculations

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

annetill, 2022-02-23:
Can you join me on slack for more details?

AnkurArohi, 2022-02-23:
Sure, how is the channel named?

annetill, 2022-02-23:
https://app.slack.com/client/TG8ALA0TB/DQAUZPHKR/thread/C01QXRH9GT1-1645606957.145159

AnkurArohi, 2022-02-23:
it seems i still need workspace

miovd, 2022-02-23:
Hello! You can join the workspace here https://join.slack.com/t/powsybl/shared_invite/zt-rzvbuzjk-nxi0boim1RKPS5PjieI0rA

annetill, 2022-02-23:
Following our conversation on slack, it seems that there is no issue related to this case. On micro grid NL case, we have only one substation and a lot of dangling lines, that will become tie lines after merging with neighbors. The issue you have, on our private case (based on IEEE 14 buses case) seems to be related to the fact that after CGMES import, the network is divided in multiple connected component. To perform a load flow on it, you have to choose parameter `LoadFlowParameters.ConnectedComponentMode.ALL`, keeping in mind that our open loadflow works if each component has agenerator controlling voltage on it.

geofjamg, 2022-03-13:
Can I close this issue?

AnkurArohi, 2022-03-13:
Yes pls