powsybl/powsybl-open-loadflow #473: Generator is discarded from power Flow calculation if target P unspecified +leading to error

AnkurArohi, 2022-03-13:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2022-03-13:
What do you mean by unspecified? zero?
Generator with a zero target p are not discarded from calculation but are discarded from slack distribution. So it means that even if network is not balanced, target will remain constante during calculation.
Are you asking to not discard from slack distribution, generators with a zero target P ?

AnkurArohi, 2022-03-13:
@geofjamg  I understand your point but there are certain cases where the generator has zero target P value because it has to be calculated/estimated from power flow algorithm, as it has a potential to produce /have a target P but only that it is not defined in cgmes or so to say it is not specified before.

Hence these generators will take part also in slack distribution or active power distribution.

Hence this is an additional advanced feature, in which the generators which do not have a predefined target P in cgmes can have one after the calculation (like after OPF set points) and its output is written in SV file after power flow.

A suggestion of how it can be executed is also explained in the ticket , it is simply a statistical analysis

vidaldid-rte, 2025-03-12:
In many uses cases for a workflow:
    - A group with target P set to 0 is considered stopped and excluded from slack distribution
    - A group with targetP outside of [pmin, pmax] is consdered starting or in any exceptional condition and excluded from slack distribution.

With this, if a group has an unknown target P and participates to slack distribution, it can be initialized with any value between minP and maxP.
However the initial value has impact on the result (it impacts the final value of that group as well as the way other groups will have to compensate). To provide a meaningful simulation result, a preprocessor should provide initial target P for the unknown groups as realistic as possible.