powsybl/powsybl-open-loadflow #474: Connected loads and generators in other CC has no calculation/power flow possibility

AnkurArohi, 2022-03-13:
* **Do you want to request a *feature* or report a *bug*?**

feature

* **What is the current behavior?**
In the current version the connected gen and loads in other CC are listed with their UUID's as a Summary from the network calculation.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**
The expected behavior is to also provide a possibility to run Power Flow in the other section of the network / islanded network,
and hence display their results too in this case they are then separate networks with separate slack buses for these  islanded networks.

the vision in future is to provide the possibility to topologically connect these islands with switches to the main network and the connection point can be para metered by the user after the topological analysis in the network.

for example Island 1 has x gens, y loads, can be connected topologically to UUID/RDFID=? 
This UUID/RDFID can be a list provided by user for the PCC's and for  each element there will be a Power Flow including the main network connected at the element UUID/RDFID specified by the user.

* **What is the motivation / use case for changing the behavior?**
This way the whole network and the islanded network can be simulated together in different topological connection scenarios, 
this makes our software robust and offers additional features depending on the user decisions.
Makes the software more fit for SCADA,
because this list of RDFID/UUID for connection points can later come from SCADA software

* **Please tell us about your environment:**
  - PowSyBl Version:  0.17.0
  - OS Version:  Linux/Ubuntu

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2022-03-13:
There is already an option for this:
```java
LoadFlowParameters parameters = new LoadFlowParameters()
        .setConnectedComponentMode(LoadFlowParameters.ConnectedComponentMode.ALL);
LoadFlow.run(network, parameters);
```

AnkurArohi, 2022-03-13:
@geofjamg Merci 
I will try it.

But, also it would be a good feature to let the user decide where should the islands be connected, I mean at what points, so that the whole network cab be simulated and faulty cgmes files /TP profile could still be worked on.
Maybe like a cgmes file validation and corrector.
So as to fully be able to approve a chmes file.

A suggestion of how this can be done is also explained in ticket.

AnkurArohi, 2022-03-14:
> There is already an option for this:
> ```java
> LoadFlowParameters parameters = new LoadFlowParameters()
>         .setConnectedComponentMode(LoadFlowParameters.ConnectedComponentMode.ALL);
> LoadFlow.run(network, parameters);
> ```

This works but then those generators are not being assigned any P setpoint anymore, they have P=0

geofjamg, 2022-03-14:
But you can do it before le LF:
```java
network.getGenerator('GEN').getTerminal().connect();

network.getLine('LINE').getTerminal1().connect();
network.getLine('LINE').getTerminal2().connect();
```

geofjamg, 2022-08-23:
I think we can close this issue, as a solution has been provided to change topology before running the LF and to calculate on all components.

closed 2022-08-23 by geofjamg