powsybl/powsybl-open-loadflow #526: Reactive power outside of reactive limits

sylvlecl, 2022-05-03:
* **Do you want to request a *feature* or report a *bug*?**

Bug.

* **What is the current behavior?**

It seems that the bus-level reactive power, when distributed on generation units at the end of the computation, can exceed their reactive limits.

See the outcome of a loadflow run, when the bus-level limit is respected, but the limit on GEN-201#2 is exceeded (20 > 16).
```
             bus_id  max_q          q
id                                   
GEN-201    VL-228_0   10.0 -10.000000
GEN-201#0  VL-228_0   10.0 -10.000000
GEN-201#1  VL-228_0   30.0 -20.896113
GEN-201#2  VL-228_0   16.0 -20.896113
```

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

(to come)

* **What is the expected behavior?**

When distributing on generation units, we should take care of keeping Q inside their respective limits.

AnkurArohi, 2022-05-04:
@sylvlecl Good point
Here the distribution should be a constraint distribution of Q respecting the limits.
It should not be a problem

Idea 💡  we can use 
Import numpy as np
If gen.q in np.arange(-maxq , maxq);
  Accept it 
Else 
  Continue redistribution 
Same for P

sylvlecl, 2022-05-04:
Actually, we found out there is already such a logic implemented:
https://github.com/powsybl/powsybl-open-loadflow/blob/9215b2c5b4d2e926fed8d278a0d56ab3d5219dcb/src/main/java/com/powsybl/openloadflow/network/impl/AbstractLfBus.java#L453

But there is a small mistake in the implementation, that @EtienneLt is fixing.

closed 2022-05-19 by annetill