powsybl/powsybl-open-loadflow #552: Support multiple generators connected to same bus controlling voltage of a different bus

geofjamg, 2022-06-13:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
If we have several generators connected to same bus but controlling voltage of different buses we have the following exception:
```
"Generators [_ee520ab7-786a-499b-a3ae-d96fb5b12596, _9064117a-2da4-46db-b1ee-f60810974a76] connected to bus '_55207717-1417-458c-9b5f-e67a3eedc46a_0' must control the voltage of the same bus".
```

* **What is the expected behavior?**
We can support this case by just adaping the LF network creation (so without any change in the equations) by just creating  `FakeLfBus` in addition to real `LfBus` for other generators connected to the bus and by linking `FakeLfBus` to real `LfBus` with a `FakeLfSwitch`. Fake or fictitious in that case means there is no equavalent object in IIDM. 

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2022-08-29:
@annetill after thinking again and prototyping the proposed solution (not working...) I am wondering if the real and clean solution is to move controller from LfBus to LfGenerator. So the VoltageControl class will reference LFGenerators as controllers. voltageControlEnabled will move to LfGenerator too. It will kind of complexify some part of the code but it will be probably cleaner as controller are LfGenerator and not LfBus. And this is exactly what we did for ShuntVoltageControl where controllers are LfShunt and not LfBus.

annetill, 2023-12-12:
Can I close this issue?

vidaldid-rte, 2025-03-12:
@geofjamg can this really be done ? If you look at the number of equations:

  - You remove one Q constraint on a bus  (or on the sum of two connected bus wich is the same) and add one variable.
  - You add two V constraints on different buses
  

So you end up with more constraints than variable. Splitting the bus would not change anything. The additional variables are linked.

geofjamg, 2025-03-12:
> [@geofjamg](https://github.com/geofjamg) can this really be done ? If you look at the number of equations:
>
> * You remove one Q constraint on a bus  (or on the sum of two connected bus wich is the [...]

Indeed not really possible to support this

closed 2025-03-12 by vidaldid-rte