powsybl/powsybl-open-loadflow #591: Graph connectivity: automatic saving

geofjamg, 2022-08-17:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
When running:
```java
var connectivity = new NaiveGraphConnectivity<>(LfBus::getNum);
for (LfBus lfBus : lfNetwork.getBuses()) {
    connectivity.addVertex(lfBus);
}
 for (LfBranch lfBranch : lfNetwork.getBranches()) {
    connectivity.addEdge(lfBranch.getBus1(), lfBranch.getBus2(), lfBranch);
}
connectivity.getNbConnectedComponents();
```

We get:
```
com.powsybl.commons.PowsyblException: Cannot compute connectivity without a saved state, please call GraphConnectivity:: startTemporaryChanges at least once beforehand
```

* **What is the expected behavior?**
So calling `startTemporaryChanges` has to be done explicitly.
Proposition: when accessing graph connectivity result (like with `getNbConnectedComponents`), we could automatically start temporary changes.

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

flo-dup, 2022-09-21:
The check done in implementations to see if a save was done beforehand can be easily removed **except for** the Even-Shiloach implementation. Several ways to deal with it for this implementation:
- do a hidden startTemporaryChanges, then the connectivity computation, and then call undoTemporaryChanges, so that the user can still change the graph before starting the temporary changes
- very similar to above, compute the level map and then invalidate it
- use jgrapht ConnectivityInspector to compute the connected set(s), and maybe throw exception if more than one connected set (not supported so far in this implementation for the temporary changes)
- throw exception only for this implementation

... or keep current behaviour?

flo-dup, 2022-09-29:
Should we close this @geofjamg ?

geofjamg, 2022-09-29:
Yes

closed 2022-09-29 by flo-dup