powsybl/powsybl-open-loadflow #594: Graph connectivity: get incremental added and removed vertices and edges

geofjamg, 2022-08-29:
* **Do you want to request a *feature* or report a *bug*?**
featue

* **What is the expected behavior?**
To ease to use of GraphConnectivity for actions simulation, we could enrich the API to expose the list of vertices and edges that have been removed and added. In actual code, there is no way to get infos on reconnected subgraph and also for disconnected subgraph we can only get the final disconnected part not the one from temporary changes (through get small components, on from component number).

```java
var connectivity = ....
connectivity.startTemporaryChanges();
connectivity.addEdge(...);
connectivity.removeEdge(...);

connectivity.getVerticesMovedToSmallComponents();
connectivity.getVerticesMovedToMainComponents();
connectivity.getEdgesMovedToSmallComponents();
connectivity.getEdgesMovedToMainComponents();

connectivity.undoTemporaryChanges();
```

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2022-08-29:
@annetill @flo-dup what do you think about this?

flo-dup, 2022-09-29:
As mentioned in PR #611, I preferred to avoid speaking of "small" components, hence the following interface:
```java
Set<V> getVerticesRemovedFromMainComponent();
Set<E> getEdgesRemovedFromMainComponent();
Set<V> getVerticesAddedToMainComponent();
Set<E> getEdgesAddedToMainComponent();
```

closed 2022-09-30 by geofjamg