powsybl/powsybl-open-loadflow #622: DC security analysis: support operator strategies

annetill, 2022-10-14:
* **Do you want to request a *feature* or report a *bug*?**

A feature.

* **What is the current behavior?**

We can only run a DC security analysis with contingencies but without any operator strategy.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**

I think that the easiest way to handle this issue is to start by supporting `LineConnectionAction`. We can use the Metrix 6 buses case as unit test. 
- First, we run a classical DC security analysis with contingencies using the sensitivity analysis API. 
- After the run, for each contingency we have to check for each operator strategy if the condition is fulfilled. If yes, we have to apply the action.
1) In case of a line déconnection, you have to that the initial network, apply the contingency and apply the action as we do for AC implementation. Run a DC loadflow and get the violations. 
2) In case of a line connection, first we have to insure that the line is inside the initial network (maybe the complex part, I could help on that) but disabled. Then you apply the contingency and the action. Run DC loadflow and get the violations.

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

obrix, 2022-10-25:
@jlbouchot After some discussion with @annetill It seems one of the step for this implementation is to move AcSecurityAnalysis.runActionSimulation into AbstractSecurityAnalysis to be usable from DcSecurityAnalysis.
We should probably also create a DcLoadFlowContext and factorize what is in common with AcLoadFlowContext.

annetill, 2024-01-09:
I close the issue because after #927, AC and DC security analysis codes are shared. We just have to continue the work of implementing remaining remedial actions.

closed 2024-01-09 by annetill