powsybl/powsybl-open-loadflow #883: `FAILED` status in case of slack distribution failure

annetill, 2023-10-17:
### Describe the current behavior

In case of slack distribution failure, we have a `CONVERGED` status (if we have the option not to throw). But it should be a `FAILED` status.

### Describe the expected behavior

`FAILED` expected and not `CONVERGED`.

jeandemanged, 2023-10-30:
@annetill @geofjamg I think we should be able to select the desired behavior though option.

Today we only have `throwsExceptionInCaseOfSlackDistributionFailure` boolean.

I propose to replace this boolean with an enum `SlackDistributionFailureBehavior`:
- `THROW`: exception is thrown if at least one SynchronousComponent cannot distribute slack (same behavior as today with `throwsExceptionInCaseOfSlackDistributionFailure` set to `true`)
- `FAIL`: SynchronousComponent-s where slack could not be distributed are flagged as `FAILED` (new behavior)
- `LEAVE_ON_SLACK_BUS`: SynchronousComponent-s where slack could not be distributed remain `CONVERGED` (eventually), undistributed slack is reported in LoadFlowResult per SynchronousComponent. (same behavior as today with `throwsExceptionInCaseOfSlackDistributionFailure` set to `false`)
- `DISTRIBUTE_ON_REFERENCE_BUS`: The reference bus is used as the last resort bus where remaining slack can always be distributed. (new behavior, in line with also https://github.com/powsybl/powsybl-core/issues/2747)

geofjamg, 2023-10-30:
Ok for me. There is just something missing for the behavior of  DISTRIBUTE_ON_REFERENCE_BUS in case of a bus without any generation. I known that it won't happen when coming from a CGMES because referencePriority that will drive the reference bus location to always be associated to a generator but more generally in OLF there is no restriction for a reference bus to be at a bus with generation (mathematically there is no need for that). So what should be do in that case ? Throws an exception to warn that DISTRIBUTE_ON_REFERENCE_BUS should be used only when reference bus has a generator?

jeandemanged, 2023-10-30:
Two possibilities:
- in mode DISTRIBUTE_ON_REFERENCE_BUS, check that provided ReferenceTerminals are only from Connectables of allowed types (most of iIDM Injection sub-classes: Gen, Load, Batteries, DanglingLines, eventually BusbarSections ...)
- alternatively, amend https://github.com/powsybl/powsybl-core/issues/2747 to allow specifying reference **injections** - instead of reference **terminals** at large. Doing this you are always sure to find an injection where to allocate remaining slack.

When DISTRIBUTE_ON_REFERENCE_BUS is not feasible (extension not provided, extension specifying a Line terminal, ...) it should behave as FAIL imo

closed 2023-11-23 by annetill