powsybl/powsybl-open-loadflow #949: Fast restart

annetill, 2024-01-09:
### Describe the current behavior

The supported features are:
- Generator update: change of targetV.
- Shunt compensator update: change of section count.
- Switch update: change open/close.

### Describe the expected behavior

We need more features:
- Open/close a branch at both sides.
- Open/close a terminal (to be precised).

bdonon, 2024-01-09:
A fast restart would be super useful to accelerate the training of machine learning models that require to run a lot of simulations with small variations.
Features mentioned in the above message would directly benefit my ongoing project, hopefully drastically decreasing the training time.
Here are some related features that I would need:

- Secondary Voltage Control extension: change of targetV.
- Ratio Tap Changers: change of targetV.

geofjamg, 2024-01-26:
> A fast restart would be super useful to accelerate the training of machine learning models that require to run a lot of simulations with small variations. Features mentioned in the above message [...]

Secondary voltage control targetV pilot point and participate of control units fast restart is done in https://github.com/powsybl/powsybl-open-loadflow/pull/966. 2 and 3 windings transformers target V fast restart is done in https://github.com/powsybl/powsybl-open-loadflow/pull/967.