powsybl/powsybl-open-loadflow #954: Target priority configurability for voltage control among generators, transformers and shunts

Hadrien-Godard, 2024-01-15:
### Describe the current behavior

Currently, in case of inconsistencies between shared generators, transformers and shunts target voltages, the target selected is chosen following a priority order. This order is hard coded: first generators, second transformers, third shunts.

### Describe the expected behavior

We want the order to be configurable at a global level.

### Describe the motivation

Feature is needed in case of inconsistencies where we prefer to rely on transformers or shunts data than on generators data for instance.

### Extra Information

A parameter can be used instead of hard coded values, but beware that those hard coded values are used elsewhere. Thus, the easiest way might be to duplicate the hard coded field and parametrize one of the duplicate.

p-arvy, 2024-01-15:
I propose to add a list of 3 integers (0,1,2, duplicating the priorities of generator, transformer and shunt controls respectively) to OpenLoadFlowParameters class in order to allow the user to configure the priority order of targets. For example, (1,0,2) would indicate a transformer target priority higher than that of generators, itself higher than that of shunts.

This order would then be added as a static parameter to the VoltageControl class (when the voltage controls of the LfNetwork are created in LfNetworkLoaderImpl), and used only to return the voltage target with highest priority when "AbstractLfBus.getHighestPriorityMainVoltageControl" method is called. 

What do you think @annetill, @geofjamg and @Hadrien-Godard ?

p-arvy, 2024-01-15:
I've just seen that it's not possible to add a list of integers as an attribute of OpenLoadFlowParameters (unless by modifying the ParameterType class on the core side, but that seems a bit cumbersome). Another way would be to duplicate the three priority parameters in the class (three integers), and create the priority list when it is added to the static VoltageControl parameter.

annetill, 2024-01-16:
> I've just seen that it's not possible to add a list of integers as an attribute of OpenLoadFlowParameters (unless by modifying the ParameterType class on the core side, but that seems a bit [...]

I don't like this kind of unexplicit parameters. Can we imagine a list of string, it is what we do for outer loops order configuration.

annetill, 2024-01-16:
> I propose to add a list of 3 integers (0,1,2, duplicating the priorities of generator, transformer and shunt controls respectively) to OpenLoadFlowParameters class in order to allow the user to [...]

I think you have to keep a way to find the highest priority main voltage control and another way to find the highest priority target voltage. I understand that is only linked to data reliability, which must not change the current logic of calculations. If you change `AbstractLfBus.getHighestPriorityMainVoltageControl`, it is much more than what you want.

p-arvy, 2024-01-17:
> I don't like this kind of unexplicit parameters. Can we imagine a list of string, it is what we do for outer loops order configuration.

Okay, let's go with a list of strings.

p-arvy, 2024-01-17:
> I think you have to keep a way to find the highest priority main voltage control and another way to find the highest priority target voltage. I understand that is only linked to data reliability, [...]

`AbstractLfBus.getHighestPriorityMainVoltageControl` returns the highest priority voltage control of a bus, and this result is currently only used to get the corresponding target (in incremental outerloops and in `AcTargetVector` class). We could specify, as parameter of the method, the order of priority to be used for sorting voltage controls, thus keeping the current logic of calculation. 

This way, voltage controls sorts with hard-coded priority order are not changed in other part of the code (e.g. in `VoltageControl.isHidden` method).

jeandemanged, 2024-03-28:
closed by #955

closed 2024-03-28 by jeandemanged