powsybl/powsybl-open-loadflow #978: Area Interchange Control as OuterLoop

jeandemanged, 2024-02-12:
### Describe the current behavior

No support for Area Interchange Control

### Describe the expected behavior

Ability to include Area Interchange Control (AIC) in OLF simulations.
The classical methods of doing AIC in a load flow are:
- Though an OuterLoop
- Directly in the Newton-Raphson

We propose as a first step to implement the OuterLoop approach only.
The feature can be enabled/disabled through OpenLoadFlowParameters.

### Describe the motivation

Area Interchange Control is a classic feature of load flows - just like simulating ratio/phase tap changers. A quick google search reveals e.g. AIC is supported by:
- [DIgSILENT PowerFactory](https://www.digsilent.de/en/load-flow-analysis.html)
- [PSS/E](https://www.linkedin.com/pulse/power-flow-analysis-solution-techniques-psse-tahir-abbas/)
- [Neplan](https://www.neplan.ch/description/load-flow/)
- [powerworld](https://www.powerworld.com/WebHelp/Content/MainDocumentation_HTML/Area_Control.htm)

Entso-e documents consider also AIC as a load flow option:
- [EMF requirements](https://eepublicdownloads.entsoe.eu/clean-documents/Network%20codes%20documents/Implementation/cacm/cgmm/European_Merging_Function_Requirements_Specification.pdf) section 4.8
- [ENTSO-E SKOS codelist](https://www.entsoe.eu/Documents/EDI/Library/ENTSO-E%20SKOS%20codelist.zip) in [EDI Library](https://www.entsoe.eu/publications/electronic-data-interchange-edi-library/) also [here](https://energy.referencedata.eu/PowerFlowSettings/e4068581-713b-4d37-88e7-bca1fd78a8f7.jsonld): `cim:PowerFlowSettings.interchangeControlEnabled": "true"`

In the PowSyBl ecosystem this can be achieved today using powsybl-entsoe/balance-adjustment module, which is a quite complicated and slow implementation. Integrating the feature directly in OLF as an OuterLoop allows for faster performance, and enables also simulating AIC in security analysis - while powsybl-entsoe/balance-adjustment approach is not suitable when simulating thousands of contingencies.

### Extra Information

Requires https://github.com/powsybl/powsybl-core/issues/2883 to be implemented first in powsybl-core/iIDM

annetill, 2024-02-21:
Agreed for outer loop + parameter.

jeandemanged, 2024-02-26:
> @jeandemanged can you please give a rough timeline on when this is expected to be implemented? thank you

Hi @amir-saadat,

I terms of timeline we expect:
- June 2024 release: area model definitions in powsybl-core/iIDM (https://github.com/powsybl/powsybl-core/issues/2883)
- (hopefully) September 2024 release: this AIC feature

PR #1055 (2024-10-09): Area Interchange Control as OuterLoop
**Please check if the PR fulfills these requirements**

- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)
- [x] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem?**

Fixes  #978 

**Does this PR introduce a breaking change or deprecate an API?**
- [x] No
closed 2024-10-09 by jeandemanged