powsybl/powsybl-open-loadflow #979: Plugin mecanism to describe custom balancing / power shifts

jeandemanged, 2024-02-12:
### Describe the current behavior

We can describe how slack distribution should be performed via the following [`LoadFlowParameters` options](https://www.powsybl.org/pages/documentation/simulation/powerflow/#available-parameters):
- `balanceType`
- `countriesToBalance`

### Describe the expected behavior

In addition to OLF implementation of above the options, some use cases require more custom-taylored power distribution rules. For example using limits on loads, using limits on generators that are different than Pmin/Pmax limits, using merit order, using different participation factors for up and down directions, etc...

We propose to provide in OLF:
- an abstraction for balancing / power shifting "rules", to be used in slack distribution (either island/synchronousComponent slack distribution, or Area InterchangeControl: #978)
- ability to provide externally to OLF the logic to perform these powershifts, leaving project specificities outside OLF.

### Describe the motivation

Multiple Entso-e use cases require the use of so-called GLSKs (Generation & Load Shit Keys), whose complexitity goes beyond existing options (proportional, merit order, shifting both generators and loads, etc...). A plugin mecanism will enable covering Entso-e use cases while leaving Entso-e specificities outside OLF.

### Extra Information

To be implemented after Area Interchange Control implementation (#978)

geofjamg, 2024-02-19:
@jeandemanged This design looks good to me to move complexity outside of OLF core. 
I would implement this balancing plugin based in GLSK in the powsybl-entsoe repository because
 - this is the rigth place for entsoe specific features and where GLSK models are.
 - powsybl-entsoe already depends on OLF

As pypowsybl already has a dependency on powsybl-entsoe we would take advantage of this feature in pypowsybl for free.

jeandemanged, 2024-02-28:
> @jeandemanged This design looks good to me to move complexity outside of OLF core.
> 
> I would implement this balancing plugin implementation based in GLSK in the powsybl-entsoe repository because
> 
> * this is the rigth place for entsoe specific features and where GLSK models are.
> * powsybl-entsoe already depends on OLF
> 
> As pypowsybl already has a dependency on powsybl-entsoe we would take advantage of this feature in pypowsybl for free.

Fully agree to do this in powsybl-entsoe and it was my mistake not to create the issue, I created https://github.com/powsybl/powsybl-entsoe/issues/137 to capture this, will also add to roadmap