powsybl/powsybl-open-loadflow #1183: Compute generator remaining margin depending on change direction

vmouradian, 2025-02-07:
### Describe the current behavior

In active power distribution, when compution participation factor of a generator with PROPORTIONAL_TO_GENERATION_REMAINING_MARGIN. The participation factor is proportional to max(0, maxP - targetP), even if the distribution should reduce targetP.

### Describe the expected behavior

Use max(0, maxP - targetP) if the mismatch to distribute is positive, and max(0, targetP - minP) if mismatch is negative.

### Describe the motivation

Customer request. (+ is more consistent)

vidaldid-rte, 2025-03-12:
@vmouradian  did PR#1189 close that issue ?

vmouradian, 2025-03-13:
Yes, #1189 adresses completely the issue. I was thinking that I should wait for merging the doc ( [core: #3326](https://github.com/powsybl/powsybl-core/pull/3326), #1201) before we close it. What is the right thing to do ?

vidaldid-rte, 2025-03-13:
@vmouradian I overloooked the tasks on core. That's right to wait for them.